bruhautomation/ESP-MQTT-JSON-Multisensor #1: Feature request - support for Dallas ds18b20 temperature sensor

marksev1, 2017-04-14:
Hi,

this Dallas ds18b20 temperature sensor is a very commonly used temperature sensor besides the DHT22, could you maybe please add support also for it?

michaelmcarthur, 2017-04-27:
Hi Mark

I have added the ds18b20 temperature sensor to the multi sensor. It works by creating a extra json  "temperaturedallas" that works alongside the DHT11/22.  Just use the following in your Home Assistant sensor.yaml

`- platform: mqtt  
  state_topic: "bruh/sensornode1"  
  name: "SN1 Temperature"  
  unit_of_measurement: "°C"  
  value_template: '{{ value_json.temperaturedallas | round(1) }}'  `

To achieve this use the code bellow. It is the multi sensor code with the dallas sensor added in. You can search for all the changes by finding the word "add". Hope this helps.

`/*
  .______   .______    __    __   __    __          ___      __    __  .___________.  ______   .___  ___.      ___   .___________. __    ______   .__   __.
  |   _  \  |   _  \  |  |  |  | |  |  |  |        /   \    |  |  |  | |           | /  __  \  |   \/   |     /   \  |           ||  |  /  __  \  |  \ |  |
  |  |_)  | |  |_)  | |  |  |  | |  |__|  |       /  ^  \   |  |  |  | `---|  |----`|  |  |  | |  \  /  |    /  ^  \ `---|  |----`|  | |  |  |  | |   \|  |
  |   _  <  |      /  |  |  |  | |   __   |      /  /_\  \  |  |  |  |     |  |     |  |  |  | |  |\/|  |   /  /_\  \    |  |     |  | |  |  |  | |  . `  |
  |  |_)  | |  |\  \-.|  `--'  | |  |  |  |     /  _____  \ |  `--'  |     |  |     |  `--'  | |  |  |  |  /  _____  \   |  |     |  | |  `--'  | |  |\   |
  |______/  | _| `.__| \______/  |__|  |__|    /__/     \__\ \______/      |__|      \______/  |__|  |__| /__/     \__\  |__|     |__|  \______/  |__| \__|

  Thanks much to @corbanmailloux for providing a great framework for implementing flash/fade with HomeAssistant https://github.com/corbanmailloux/esp-mqtt-rgb-led

  To use this code you will need the following dependancies: 
  
  - Support for the ESP8266 boards. 
        - You can add it to the board manager by going to File -> Preference and pasting http://arduino.esp8266.com/stable/package_esp8266com_index.json into the Additional Board Managers URL field.
        - Next, download the ESP8266 dependancies by going to Tools -> Board -> Board Manager and searching for ESP8266 and installing it.
  
  - You will also need to download the follow libraries by going to Sketch -> Include Libraries -> Manage Libraries
      - DHT sensor library 
      - Adfruit unified sensor
      - PubSubClient
      - ArduinoJSON

*/

#include <ESP8266WiFi.h>
#include <DHT.h>
#include <PubSubClient.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <ArduinoJson.h>

// Add this here
#include <OneWire.h> 
#include <DallasTemperature.h>
//

/************ WIFI and MQTT INFORMATION (CHANGE THESE FOR YOUR SETUP) ******************/
#define wifi_ssid "YourSSID" //type your WIFI information inside the quotes
#define wifi_password "YourWIFIpassword"
#define mqtt_server "your.mqtt.server.ip"
#define mqtt_user "yourMQTTusername" 
#define mqtt_password "yourMQTTpassword"
#define mqtt_port 1883

/************* MQTT TOPICS (change these topics as you wish)  **************************/
#define light_state_topic "bruh/sensornode1"
#define light_set_topic "bruh/sensornode1/set"

const char* on_cmd = "ON";
const char* off_cmd = "OFF";

/**************************** FOR OTA **************************************************/
#define SENSORNAME "sensornode1"
#define OTApassword "bruh"
int OTAport = 8266;

/**************************** PIN DEFINITIONS ********************************************/
[... 509 lines cut ...]
   reaches the step size appropriate for one of the
   colors, it increases or decreases the value of that color by 1.
   (R, G, and B are each calculated separately.)
*/
int calculateVal(int step, int val, int i) {
  if ((step) && i % step == 0) { // If step is non-zero and its time to change a value,
    if (step > 0) {              //   increment the value if step is positive...
      val += 1;
    }
    else if (step < 0) {         //   ...or decrement it if step is negative
      val -= 1;
    }
  }

  // Defensive driving: make sure val stays in the range 0-255
  if (val > 255) {
    val = 255;
  }
  else if (val < 0) {
    val = 0;
  }

  return val;
}
`

corbanmailloux, 2017-04-27:
@michaelmcarthur When pasting code in GitHub, wrap the code in three backticks (```).

Like:

\```
code
\```

michaelmcarthur, 2017-04-27:
Thanks for the advice @corbanmailloux  its my first time using github. Updated post

marksev1, 2017-05-09:
Are you gonna make a PR to the main code? :)

michaelmcarthur, 2017-05-09:
This code answers the question of adding ds18b20 sensors, but it isn't polished enough. I'm working on a way to create add on's to the code. This will allow any sensor to be added without changing the core code every time a sensor or switch is added. An example of this is @1technophile OpenMQTTGateway. It makes it very easy to add your own code as add on's.

marksev1, 2017-05-10:
Good plan. I would envision only maybe those new "radar sensors" which are like better than PIRs. To be another viable addition.

michaelmcarthur, 2017-05-11:
Those "radar sensors"  look really great. I would love to see a working DIY Arduino version of them. I have 433mhz transmitter and IR added to Bens code in the same way as I did the ds18b20. I find them really useful in my setup.

marksev1, 2017-05-11:
Yeah I'm also waiting for that...So if I find you some nice code, you will integrate those radar sensors too? :D ...Good I see you are mixing in a bit of OpenGateway also in right :), cool cool.

bruhautomation, 2017-05-11:
That's some great code. Thanks much for sharing @michaelmcarthur. 

@marksev1, which radar sensor are you looking at? I tried several when working on this code, but I was having issues with RF interference. I'm still working to figure out the right series of decoupling capacitors to isolate the noise. Otherwise, I'll have to switch to an RF protocol and use a MQTT->RF bridge via something like MQTTOpenGateway. 

Cheers,
Ben

marksev1, 2017-05-12:
@bruhautomation  this one: RCWL-0516, its very cheap, and works omnidirectionally (it even works through some not so thick walls), as opposed to PIR sensors. 
This guy got it working with a esp8266: http://tech.scargill.net/microwave-for-the-weekend/

michaelmcarthur, 2017-05-12:
@marksev1 The RCWL-0516 motion sensor looks amazing, thank you for sharing. I have ordered one today should be with me next week. You should be able to hook it right up to the multi sensor. The RCWL-0516 sends a high signal when motion is detected so you should be able to remove the pir sensor from the multi sensor and add this one.

bruhautomation, 2017-05-12:
@marksev1  @michaelmcarthur Ah, gotcha! yeah, it's a great sensor. It works wonderfully on an Arduino. A couple of downsides so far in my early testing though. 

1) On the NodeMCU it  is very susceptible to false positives. Long header wires reduce this. Shielding and decoupling capacitors could fix this, but I haven't found the magic sauce yet. 

2) It's detection area is essentially a 360 degree sphere. When I used this on the second floor of my house, it detected motion when I walked around downstairs. That may not be an issue for some, but for room level automation this posed a challenge.

I'm curious to see how your attempts go! Let me know if you get it working. Cheers!