bruhautomation/ESP-MQTT-JSON-Multisensor #2: Feature request - Support for a NeoPixel Ring - 12 x 5050 RGBW LEDs

hinrg, 2017-04-26:
Hi

Love your multi sensor :) do you think we can get support for a neo pixel ring rather then the LED. [https://littlebirdelectronics.com.au/collections/neopixel-rings/products/neopixel-ring-12-x-5050-rgbw-leds-w-integrated-drivers-cool-white-6000k](url) I'm working on a 3D printed case which i'll share once i'm done would be cool to have a led ring on the top like the google home ;)

[image]

Cheers,
J

bruhautomation, 2017-04-26:
Sweet enclosure! Looks great. 

I would love to support the FastLED library (which supports neopixel rings) on this sketch. My long term goal is to merge my multisensor and led sketches into a single sketch. People can use whatever features they want from either. 

This is on my list once I wrap up my blinds project. Looking forward to knocking that out soon. 

Cheers,
Ben

hinrg, 2017-04-27:
Nice exactly what i was thinking!!!

msenebald, 2017-04-30:
i will be testing too 👍

Wilco89, 2017-06-07:
What is the state of your case? it looks very nice

cooljimy84, 2018-03-21:
Got myself a Neo Pixel ring, and a touch switch for local control.
If i get anywhere, i'll link here