bruhautomation/ESP-MQTT-JSON-Multisensor #7: MQTT_MAX_PACKET_SIZE not being recognized

Terr4, 2017-05-07:
Hi there,

I have build your multisensor and have now spend quite some hours figuring out why my MQTT broker did not receive any JSON messages from it, even though it was successfully authenticated and registered and the ESP could send MQTT test messages.

It seems the parameter MQTT_MAX_PACKET_SIZE does not get applied in my case and no JSON message was sent therefore as they are longer than 128 bytes. For my testing I used your code 1:1, except for the credentials and broker server ip.

The only solution was to change the MQTT_MAX_PACKET_SIZE parameter in the library file "PubSubClient.h" itself.

I see that 8 days ago they made some changes in that area to actually make the size configurable in an INO script, however so far the mentioned "setBufferSize()" method is not available in my library version:
https://github.com/knolleary/pubsubclient/pull/282/commits/5b23a97fb00f73c8f8bd4a0bcee9576ae62c8d5c

br
Stefan

tomdee, 2017-05-14:
@Terr4 thanks for raising this. I spent a good few hours trying to work out why I wasn't getting any MQTT messages as this turned out to be the problem

bruhautomation, 2017-05-23:
Interesting. That's very strange. Sorry about the issue. I'll add a comment to the code and check into this and see if there is anything I can do. Thanks much for catching it!

closed 2017-05-23 by bruhautomation

cooljimy84, 2017-07-03:
I'm having this kinda issue, the ESP just isn't publishing any thing, but can receive stuff. I've sent the MQTT_MAX_PACKET_SIZE in the PubSubclient.h from 128 to 512, 1024 are reuploaded, doesn't see to help....
Where abouts would i set the buffersize in the config Bruh has uploaded ?