bruhautomation/ESP-MQTT-JSON-Multisensor #8: MQTT not functional after Wifi disconnect

knutella, 2017-05-12:
About once a day my sensor would just stop being able to send or receive data over MQTT.  I could still ping the MCU.  The serial monitor showed nothing.

I was able to replicate the issue either by simply rebooting my router or using the Wifi MAC filter to temporarily block the MCU.  Afterwards the MCU would reconnect to wifi (as evidenced by the ping) but no longer function for MQTT send/receive.

TheFitzZZ, 2017-05-16:
I may experience the same issue. I have six sensors around the house and yesterday I noticed that two weren't sending updates anymore for no reason. Maybe we need some kind of automatic restart or so?

knutella, 2017-05-16:
I wrote a fix for this issue and submitted it, its been working reliably for about a week now.

[sketch_may04a.txt](https://github.com/bruhautomation/ESP-MQTT-JSON-Multisensor/files/1004853/sketch_may04a.txt)

bruhautomation, 2017-05-23:
Sorry about the issue! Knutella has a solid fix for this which I will merge ASAP. Thanks much for reporting it and for the fix @knutella!

TheFitzZZ, 2017-05-24:
Can confirm that this solves this issue perfectly, have been testing it since @knutella posted the fix.

bruhautomation, 2017-05-29:
I just merged this code into the repo so the fix should now be live. Thanks for the patch @knutella! Cheers!

closed 2017-05-29 by bruhautomation