bruhautomation/ESP-MQTT-JSON-Multisensor #9: PIR Oddities!

jonathanweinberg, 2017-05-16:
Great tutorial / project! Having a slight issue with the PIR portion.

I initially wired everything up with a LDR (waiting on a TEMP6000), and an HC-SR501, and couldn't for the life of me get it to work, it was either all motion, or no motion. It was a cheapie and I figured it was the issue you described.

After long wait, my fancy AM312's got here! **Same thing.**

So I pulled out my analog meter, and saw very very little between Red (VIN) and Black (G) for the PIR sensor. So I move Red (labeled VIN) from it's place by the Black Ground, and up to the 3V by D4. Seems to be fine!

After 2-3 seconds it does show as standby if there's no active motion, but that's functionality to expect from the AM312 as I understand it.

Thoughts? Did I do something silly?

bruhautomation, 2017-05-23:
Sorry about the issue and my delay getting back to you. That's interesting about the issues with the AM312. 

So, power the PIR from the 3V V pin near D4 made the sensor work for you? In normal operation, the PIR signal pin should remain low until it detects motion. It'll pull high for 2-3 seconds and then go low again automatically. Is this the behavior you're seeing?

bruhautomation, 2017-05-29:
Closing this due to inactivity. Feel free to reopen if you need to.

jonathanweinberg, 2017-07-05:
Reopening! :) Other similar issues #25 and #19 

So, after looking at this more closely, I'm not sure why using VIN + GND + D5 would work. Keep in mind, I'm still trying to learn basic electronics.

The VIN pin, is literally as far as I know, [current into the NodeMCU voltage regulator](http://henrysbench.capnfatz.com/henrys-bench/arduino-projects-tips-and-more/powering-the-esp-12e-nodemcu-development-board/). It's how you would power the unit if not using a USB power adapter.

So we have `GND`, `Voltage into the NodeMCU`, and a `Digital Pin`. `GND` and `VIN` are both "sinks" of current. [In the Arduino sketch, the PIR pin `D5` is an input](https://github.com/bruhautomation/ESP-MQTT-JSON-Multisensor/blob/75fc667fd1309191c01965b2b5c3fb4d9fcb9c8d/bruh_mqtt_multisensor_github/bruh_mqtt_multisensor_github.ino#L145)

Does my logic make sense or am I full of crazy talk here?

seanfilyaw, 2017-07-09:
I have wired according to the Diagram, in my instance I have a constant motion detected being displayed, on the Uxcell Mini Pyroelectric PIR Sensor I have the r2 red(vin) c3 yellow(signal) and r1 black(ground) I guess my question is do I have this wired correctly, It seems like it is constantly pulling high

jonathanweinberg, 2017-07-10:
@seanfilyaw - Question for you. Do you have a TEMT6000 in use? Or are you using an LDR?

seanfilyaw, 2017-07-10:
LDR