bruhautomation/ESP-MQTT-JSON-Multisensor #14: 'D1' was not declared in this scope

DoomzKid, 2017-05-24:
I get this error when, Any Idea how to fix this ?

-------------------------------------------------------------
Multisensor:37: error: 'D1' was not declared in this scope

 const int redPin = D1;

                    ^

Multisensor:38: error: 'D2' was not declared in this scope

 const int greenPin = D2;

                      ^

Multisensor:39: error: 'D3' was not declared in this scope

 const int bluePin = D3;

                     ^

Multisensor:41: error: 'D7' was not declared in this scope

 #define DHTPIN    D7

                   ^

C:\Users\*****\Documents\Arduino\Multisensor\Multisensor.ino:105:9: note: in expansion of macro 'DHTPIN'

 DHT dht(DHTPIN, DHTTYPE);

         ^

C:\Users\****\Documents\Arduino\Multisensor\Multisensor.ino: In function 'void setup()':

Multisensor:40: error: 'D5' was not declared in this scope

 #define PIRPIN    D5

                   ^

C:\Users\*****\Documents\Arduino\Multisensor\Multisensor.ino:114:11: note: in expansion of macro 'PIRPIN'

   pinMode(PIRPIN, INPUT);

           ^

Multisensor:41: error: 'D7' was not declared in this scope

 #define DHTPIN    D7

                   ^

C:\Users\******\Documents\Arduino\Multisensor\Multisensor.ino:115:11: note: in expansion of macro 'DHTPIN'

   pinMode(DHTPIN, INPUT);

           ^

C:\Users\*****\Documents\Arduino\Multisensor\Multisensor.ino: In function 'void loop()':

Multisensor:40: error: 'D5' was not declared in this scope

 #define PIRPIN    D5

                   ^

C:\Users\*****\Documents\Arduino\Multisensor\Multisensor.ino:397:28: note: in expansion of macro 'PIRPIN'

     pirValue = digitalRead(PIRPIN); //read state of the

                            ^

exit status 1
'D1' was not declared in this scope

TheFitzZZ, 2017-05-24:
Please make sure you selected the right board.

DoomzKid, 2017-05-24:
Thanks !, 

I was using the video on youtube, 
Testing the file, But a bit further in the video. 
Doas he the settings of the board. after that the code "D1" Error wiln't show, 

Now i am only stuck at Serial Montitor with the message> 

failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds... ... ..

bruhautomation, 2017-05-29:
It's likely an issue with the MQTT server or credentials you are using. Verify your MQTT adress and the username/password info and try again. Cheers!

bbrendon, 2017-06-08:
I picked this board and I think my problem is fixed.

[image]

timmo001, 2017-10-16:
If that fixed your issue, make sure you hit close issue on this thread

Philje123, 2017-12-04:
I too had this "issue".

I was using an ESP-01 board and put D0 and D2. But they're not defined as data pins on the ESP-01. So if this is the case you have to just put 0 and 2 for these boards.

mmilton62, 2018-07-29:
Tks. I was using ESP8266 and had the same problem. Changed to ESP-12 1.0, as sudgested by bbrandon, and it all workrd.