bruhautomation/ESP-MQTT-JSON-Multisensor #15: Getting very inaccurate temperature readings

TheFitzZZ, 2017-05-28:
Hey guys,

I have six of the sensors around the house, some also where already other sensors exist, so  I can compare the results. Something I don't really need another sensor for is the fact, that the remperature is off by some extend. It's always way too high by 5-10°C - same error with °F btw. 

Any ideas what could cause this? It makes them practially useless. 

Thanks

tariquesani, 2017-05-28:
Do you by any chance have DHT11 rather than DHT22

TheFitzZZ, 2017-05-28:
No, but I think I see the issue now: the case. I think it heats itself up and the ventilation by the few left-outs isn't enough by any chance to get proper readings.

How I know this? I aimed a little USB ventilator at the box to get some airflow. And voila, the temperature seems much more likely to be correct: went from 31°C to 24° - which seems reasonable.

I will verify further, design a new case and make a pull request for it once I've tested it ... will take some time though :-)

bruhautomation, 2017-05-29:
Yeah, in doing some testing with this tonight. It does appear that the case is causing the heating issues. Sorry about that! You could add an offset in the code, but that's not a great solution. I'm going to redesign the case and move the sensor further from the board. That should help. 

@TheFitzZZ, if you come up with a new case design, too. I'd be happy to post the link or host the STL in the repo if you want!

Cheers!

gillrishi, 2017-05-30:
Hi
I am using dht11 sensor.
 it shows temprature 987 c and humidity 1098%.
Can you help me out to fix this issue?
Thanks

tariquesani, 2017-05-30:
@gillrishi get a dht22

gillrishi, 2017-05-30:
Ok..
Thanks for your help

bruhautomation, 2017-06-02:
Yeah DHT11 sensors aren't as great. If you use a DHT11, make sure you set line 73 to

```
#define DHTTYPE   DHT11
```

Cheers!

cbrherms, 2017-12-03:
Looks like the V2 case might be doing something similar though. i'm getting about 5C difference even on the new case.

martinjuhasz, 2018-01-09:
Same here, its related to the case. Having the sensor some cm outside of the box shows me the correct temp.

Any ideas what we can do about this?