bruhautomation/ESP-MQTT-JSON-Multisensor #16: Compilation issue on Arduino 1.8.2 on Linux !

vincegre, 2017-06-01:
Following your great youtube tutorial for setup of the MQTT client with ESP8266 boards !
I have copy/pasted your code from GitHub in Arduino IDE 1.8.2 running on Ubuntu 16.04.2 (I have added all libraries as indicated in your video). I then edited the few lones for Wifi setup, network and MQTT server.
When I try to compile code I have an error:
`Arduino : 1.8.2 (Linux), Carte : "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

MQTT-ESP2866:65: error: 'D1' was not declared in this scope
 const int redPin = D1;
                    ^
MQTT-ESP2866:66: error: 'D2' was not declared in this scope
 const int greenPin = D2;
                      ^
MQTT-ESP2866:67: error: 'D3' was not declared in this scope
 const int bluePin = D3;
                     ^
MQTT-ESP2866:69: error: 'D7' was not declared in this scope
 #define DHTPIN    D7
                   ^
/home/vincen/Bureau/MQTT-ESP2866/MQTT-ESP2866.ino:133:9: note: in expansion of macro 'DHTPIN'
 DHT dht(DHTPIN, DHTTYPE);
         ^
/home/vincen/Bureau/MQTT-ESP2866/MQTT-ESP2866.ino: In function 'void setup()':
MQTT-ESP2866:68: error: 'D5' was not declared in this scope
 #define PIRPIN    D5
                   ^
/home/vincen/Bureau/MQTT-ESP2866/MQTT-ESP2866.ino:142:11: note: in expansion of macro 'PIRPIN'
   pinMode(PIRPIN, INPUT);
           ^
MQTT-ESP2866:69: error: 'D7' was not declared in this scope
 #define DHTPIN    D7
                   ^
/home/vincen/Bureau/MQTT-ESP2866/MQTT-ESP2866.ino:143:11: note: in expansion of macro 'DHTPIN'
   pinMode(DHTPIN, INPUT);
           ^
/home/vincen/Bureau/MQTT-ESP2866/MQTT-ESP2866.ino: In function 'void loop()':
MQTT-ESP2866:68: error: 'D5' was not declared in this scope
 #define PIRPIN    D5
                   ^
/home/vincen/Bureau/MQTT-ESP2866/MQTT-ESP2866.ino:424:28: note: in expansion of macro 'PIRPIN'
     pirValue = digitalRead(PIRPIN); //read state of the
                            ^
exit status 1
'D1' was not declared in this scope`

Missed something ? I just checked your video but it looks like I have missed none steps :(

Thanks

Vincèn

bruhautomation, 2017-06-02:
Are you using a NodeMCU 1.0 board? You might try selecting that in the board manager (if you haven't yet) and see if that helps. Sorry about the issues!

vincegre, 2017-06-02:
@bruhautomation Unhappy not ! I use the ESP board as I follow your tutorial !

bruhautomation, 2017-06-02:
Looking at your error log there is appears you have the ESP8266 board selected? 
```
Arduino : 1.8.2 (Linux), Carte : "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None" 
```

You should select NodeMCU 1.0.

vincegre, 2017-06-02:
@bruhautomation ah yep works fine and compile with success ;) sorry for mistake ! and thanks for help !

closed 2017-06-02 by vincegre