bruhautomation/ESP-MQTT-JSON-Multisensor #17: Compliation issues

amitubale, 2017-06-08:
Hi anyone able to compile the code with the new versions o the packages ?.

 i tries for a whole day and gave up. i kept getting the IRemote..error. I removed the folder but no luck. since i waited @a month had to get the sensor working so i have espeasy on it now and got the 3 sensors working. (not sure how to get the color led configured on esp..so i need to get Ben's code compiled.. 

I iwll try it again and post the error.
let me know if you are bale to compile with the new pakages ? thanks

amitubale, 2017-06-09:
i was able to compile after adding the DHT.h lib and had to add #include <Adafruit_Sensor.h> above dht.h
and it connects to Wifi too. took me while to realize there i no webserver in the code(or did i miss another lib ?)

tariquesani, 2017-06-09:
No, there is no webserver but you can see output in the serial monitor. This works best with Home Assistant

amitubale, 2017-06-09:
awesome..lean code..thanks

ArjenWols, 2017-07-25:
Compilation error on Adafruit_Sensor.h
Solved with this:   https://github.com/adafruit/DHT-sensor-library/issues/62
Regards
Arjen