bruhautomation/ESP-MQTT-JSON-Multisensor #19: No Change in PIR Motion Detected State

epleypa, 2017-06-09:
I'm running into an issue where there does not seem to be any change from "Motion Detected" back to "standby." Since I'm new to the ESP and PIR, any tips on troubleshooting? I've tried two different PIR modules with no changes, both report motion as soon as the board has booted. The code seems to be working as I get standby if I unplug Vin from the PIR. Could be bad hardware, but I'm unsure. Thanks!

tariquesani, 2017-06-09:
Which PIR are you using? Does it have setting for delay and repeater?

epleypa, 2017-06-09:
I am using the mini PIR AM312 from Ben's Aliexpress link: https://www.aliexpress.com/item/Mini-IR-Pyroelectric-Infrared-PIR-Motion-Human-Sensor-Automatic-Detector-Module-high-reliability-12mm-x-25mm/32749737125.html. I don't think it has a setting for delay or repeat from what I can tell.

epleypa, 2017-06-10:
Looks like it was a bad PIR and Vin on the ESP8266, just got it working on another set

closed 2017-06-10 by epleypa