bruhautomation/ESP-MQTT-JSON-Multisensor #20: Bad data send when reboot

wanghuangjie, 2017-06-09:
{"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"humidity":"0.00","motion":"motion detected","ldr":"0","temperature":"0.00"}
{"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"humidity":"0.00","motion":"motion detected","ldr":"757","temperature":"0.00"}
{"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"humidity":"0.00","motion":"motion detected","ldr":"757","temperature":"86.00"}

Multisensor will send "0" when reboot, please help if there is any solution .

jumpalottahigh, 2017-06-25:
Hi. I seem to have a similar issues - the multisensor does not report humidity for some reason. Everything else works fine and I know it's not a faulty sensor, since I tested it separately, and also tested with a second DHT22 sensor. Will look into this today and let you know if I figure something out.

[image]

jumpalottahigh, 2017-06-25:
Double check your code, I had a spelling mistake (although the program would still compile).
Had a `float newHumValue = dht.read();` which should have been float `newHumValue = dht.readHumidity();`. No idea how that happened, but everything works like a charm.