bruhautomation/ESP-MQTT-JSON-Multisensor #22: LDR all over the place

melsom, 2017-06-24:
Hello,

Is there any way to correct this? The sensor i stationary but the values seems to be up/down alot even though the median seems to be correct, looking at the 24h logs.

[image]

[image]

cooljimy84, 2017-07-03:
I get this when someone walks in front of the sensor. I was thinking of using a diffuser on top of it to try and level it out a bit.
I also have a 30 minute window where the sun bounces off cars going past and they reflect into the flat causing it to read higher for a second or so....

jonathanweinberg, 2017-07-05:
Also, if you're using an LDR, `lux` is an arbitrary thing in that scale, you aren't getting true `lux` values.

ntalekt, 2017-07-09:
I got this too and realized it was my ceiling fan throwing shadows all over the place. I moved the sensor to a place that isn't affected and mounted in an enclosure and it's been rock solid.

Jizar07, 2017-11-19:
I got a different issue , my sensor works backwards, if its dark the numbers goes up to about 600 lux, if is bright is goes down to 33 lux. , I got the 4 pin sensor and got signal pin on AO (analog output) instead of DO (digital output). already tried switch AO and DO, but got the same result. any help would be greatly appreciated.

cooljimy84, 2017-11-21:
correct (well for me as well) so darkness for my sensor is 900 LUX and bright light is 20 odd.

rdbahm, 2017-12-08:
@Jizar07 At least with an LDR, this happens if  your ground and 3v are swapped.

mikewebb70, 2019-02-02:
> @mikewebb70 - you should share that template :)

You are right I should have, it's nothing special.  Here is my light sensor Home Assistant yaml entry

```
- platform: mqtt
  name: "flux office"
  state_topic: "tele/labmcu/SENSOR"
  unit_of_measurement: "%"
  #value_template: "{{ value_json['ANALOG'].A0 }}"
  value_template: "{{ (value_json['ANALOG'].A0|float/1024*100|float)|round(0) }}"
  availability_topic: "tele/labmcu/LWT"
  payload_available: "Online"
  payload_not_available: "Offline"
```