bruhautomation/ESP-MQTT-JSON-Multisensor #28: Temp in C not F?

markrennie71, 2017-07-07:
Hello

This is my 1st project like this and I love it thank you. Since I am English, how do I get it to display the temperature in C not F please?

Cheers
Mark

AaronLayton, 2017-07-25:
Hi @markrennie71 

Easy - just remove the `true` from this line
https://github.com/bruhautomation/ESP-MQTT-JSON-Multisensor/blob/master/bruh_mqtt_multisensor_github/bruh_mqtt_multisensor_github.ino#L449

so it should read

```
float newTempValue = dht.readTemperature();
```

nagash, 2017-07-26:
I believe changing the recommended line of code causes the "Real Feel" value to be incorrect as the real feel algorithm uses the same temp value but is hard coded against F.

FWIW, using home assistant, I left the value as F and noted that the input was F in the sensor config. I've got my units set to metric in home assistant config, and it "just worked". I get my values in Home Assistant in celsius and when I write automations I also specify celsius values.

markrennie71, 2017-07-26:
thanks guys, yeah I already found that and found the real free is incorrect. They are only for my loft and garage, so real free isn't required for me :)

AaronLayton, 2017-07-26:
Yeah good points 🔼 
Maybe that needs to be put into a setting so it can be easily changed