bruhautomation/ESP-MQTT-JSON-Multisensor #29: LDR Time Series

ntalekt, 2017-07-09:
Anyone else having this weird time series issue? The lights were off (0 LUX) until 1:27:33PM when I turned them on for a little bit. Instead of the time series being 0 for the time period before I turned on the light there is a diagonal line. Almost like HA isn't logging the 0 data points.

[image]

AaronLayton, 2017-07-25:
I have not had this, although, my LUX seems to be backwards - so I get a higher value when it's dark

kernelheap, 2017-08-15:
my LUX is backwards too. If I throw the iphone flashlight on it, it indicates 80 lux. Otherwise it says 565 lux. Not sure whats wrong.

kwinck, 2017-08-29:
HA does not update if the new value is the same as the previous value. That causes the spike. Try to append 
`force_update: true`
the the sensor definition.

https://community.home-assistant.io/t/force-update-settings-for-all-components-from-configuration-yaml/2198

rdbahm, 2017-10-24:
@kernelheap Reverse your VCC and GND lines.

Suggest this issue be closed for this repo as it's a Home Assistant quirk, not a problem with the sensor.

ntalekt, 2017-10-24:
@rdbahm Agreed. Thanks!

closed 2017-10-24 by ntalekt