bruhautomation/ESP-MQTT-JSON-Multisensor #33: Incorrect Temperature Readings

bachoo786, 2017-07-25:
Hi thanks for the amazing sensor diy.

I have followed everything as per the instructions but my dht22 sensor gives me incorrect temperature readings by 5 to degrees more than the current room temperature.  I have changed the dht22 sensor hoping it was faulty but I get the same problem with a new dht22 sensor. 

Has anyone come across this issue before?

Any help will be appreciated thanks.

DanielXYZ2000, 2017-10-15:
According to the datasheet the accuracy is <+-0.5°C. Large deviations could be based on the sensor position (housing, light exposure, convection,...) or maybe the sensor is defect?

mr-sneezy, 2018-04-26:
This could be from heat soaking from the NodeMCU board...
If the DHT22 is close to the WiFi RF chip or the 5V-3.3V regulator, especially mounted above it, the small amount of heat is enough to raise the temp of the sensor by as much as 5 degrees. Found out the hard way myself.
Also, it pays to only sample the DHT22 infrequently as possible, as it self heats while doing the conversions. They recommend no more than about once in 5 seconds IIRC. Can't see why the sensornode would need to be any more than every 15 seconds or more...

soppilif, 2019-03-18:
im experiencing similar issues.
Before using the enclosure, DHT22 was giving me 1 correct reading out of 5. That is:
one reading was -10 C , next reading was the correct one 21 C and next ones same as -10 C.
Similar with humidity... 
Once i used the enclosure i was getting stable readings of 12 C temperature and 22% humidity which again were incorrect.
My enclosure was a Pringles box dropped to its minimun possible height.
I removed the DHT22 at the moment until i find a solution.
Any ideas?

mr-sneezy, 2019-03-19:
That's not the same issue, such a big deviation in both directions sounds like a power supply issue  (to the sensor or the NodeMCU) . Try a large (470uF or bigger) capacitor on the 5V supply rail and ground, then if that makes not difference maybe check for wire or pin connection problems.

soppilif, 2019-03-22:
> That's not the same issue, such a big deviation in both directions sounds like a power supply issue (to the sensor or the NodeMCU) . Try a large (470uF or bigger) capacitor on the 5V supply rail and ground, then if that makes no difference maybe check for wire or pin connection problems.

tried different pins and jumpers but same behaviour. The moment i unplug and replug the jumpers one or two readings are correct and then it gets faulty.
Should i get another DHT? I cant solder capacitors.
PIR and LDR work fine.

mr-sneezy, 2019-03-22:
Do you have an Arduino board at all (Nano, Mini etc will do), if so wire it to that and try any test code on Git via the serial terminal and see if it reads right to prove the DHT is OK.
BTW, it's my understanding that Bruh has abandoned this code and now doing same stuff via a new system.