bruhautomation/ESP-MQTT-JSON-Multisensor #36: temperature and heatIndex values on first boot up

ericstewart22, 2017-08-26:
Hello,

Whenever I unplug my sensor or it loses power, the temperature and heatIndex values default to 0 and get sent to my MQTT broker which then makes its way into Home Assistant and results in a spike in the charts. 

Here's an example of what it looks like:
[image]

I tried adjusting the calibrationTime variable but that didn't work. What would be the recommended way to fix this?  Is it as simple as using a counter to disregard the first few MQTT values? If it is then maybe I can try and fix it myself, but I thought there might be a better way.

knutella, 2017-08-26:
I added this to the SendState void function.  I basically doesn't send if its below 1.  I'm sure something more elegant like just sending a NULL would be better but this works for me

if (tempValue<1) {
        Serial.println("less than 1");
      }
  else {
  Serial.println(buffer);
  client.publish(light_state_topic, buffer, true);

kwinck, 2017-08-29:
My take on that issue is to have a timer, that ensures that it doesn't send any values the first 30 sec. or so, ie. until the calibration period is done.

ericstewart22, 2017-08-31:
Thanks for the input, I'm going to use the tempValue < 1 for now but I'll try my hand at using a timer when  I have more time.

lassivv, 2017-10-23:
Same problem and try add capacitor on power line and better 5volt supply power, nothing help not know why sensor restart/softstart every day random couple times.

After that i add this code in sendstate function and everything works perfectly after that. Think every tips/hints here work too:

byte countSentStates = 0;

  if(countSentStates <= 2)
  {
    countSentStates++;
    return;
  }

mr-sneezy, 2018-04-26:
lassivv, I get these dropouts also, on two sensornodes. It seems to be a random reset from something unknown. It's not power related as I've covered that through elimination and also adding large fiters caps (1800uF) to the 5V input of the NodeMCU's just in case. 
My graphs of 24 data look very like yours with data dropouts to below zero on temp and humidity when these events occur. 
When it does this the RGB LED also goes to blue and stays that way till I send a "White' colour to it and then 'Off' via HA Light component GUI.