bruhautomation/ESP-MQTT-JSON-Multisensor #38: Blue light on sensor startup

sjorsjes, 2017-09-26:
I've got 7 BRUH multisensors up and running and I love them and I have 0 problems with them so far!

The only thing that bothers me is that I can't get rid of the blue light. It lights up when I'm connecting it to power but also on, which feels, random times. I could not find anything about this in the code so I was hoping you could push me in the right direction.
I noticed some people in the youtube comments have the same issues.

Thanks and keep up the awesome work!

knutella, 2017-09-26:
I had the same issue, I added this to the code to make it go away

void setup() {

  Serial.begin(115200);

  pinMode(PIRPIN, INPUT);
  pinMode(DHTPIN, INPUT);
  pinMode(LDRPIN, INPUT);

  **//new code to turn LED on then off
  analogWrite(redPin, 1);
  analogWrite(greenPin, 1);
  analogWrite(bluePin, 1);
  analogWrite(redPin, 0);
  analogWrite(greenPin, 0);
  analogWrite(bluePin, 0);**
  
  delay(10);

mr-sneezy, 2018-04-26:
Yes among other issues my blue LED goes on randomly or perhaps also after a reset. 
I'm just a newbie at Arduino C coding, can you tell me why you declare the RGB pins as inputs and then toggle them as output ?

BertrumUK, 2018-11-05:
You can also stop it by moving the Blue LED leg from D3 to D4 and changing the code to match.

I like the affect as it lets me know if there is an issue with Hassio or MQTT that I need to look into.