bruhautomation/ESP-MQTT-JSON-Multisensor #41: Random 0values on HA logging

lassivv, 2017-10-15:
Build sensor and leave led hardware out. Sensors readings are perfect and seems very accurate.

But randomly couple times of days, every sensor take 0value (temp, humidity, brighness)

Here example picture:
https://i.imgur.com/dSYydNj.png

lassivv, 2017-10-22:
Anyone have same type problems or do i only have them :) 

Now i have couple sensor. Other is this wiring method what this page shows. Other is totally planned circuit board. 

Wiring method here still keep giving that 0values 3-6times per day.

My circuit board model gives only one that kind 0values on day. Circuit board model is longer distance to wlan router than original model. Bad wlan signal is not that problem then.

....

One solution maybe if you can take of that 0values when bruh multisensor arduino soft boot. I think sensor do boot and because that i get that 0values. (because if i watching serial monitor on sensor everytime when i boot that that gives couple times 0values before that gives right values.

Hope someone have same problems and can give some tips.

knutella, 2017-10-23:
This looks like the same issue as https://github.com/bruhautomation/ESP-MQTT-JSON-Multisensor/issues/36

lassivv, 2017-10-23:
Confirm that, just some reason not see that issue, when i do my issue number.