bruhautomation/ESP-MQTT-JSON-Multisensor #44: I don't see any MQTT messages in a client.

ravensteijn, 2017-10-26:
Hi,
I Like this multisensor a lot! I have one problem. ik don't see the messages in an MQTT client. like MQTT spy. When i subscribe to topic # i don't get any messages. I have owntracks on my phone and that works great in HASSIO mosquitto.

Can anybody help what to look for?

This is the serial monitor in arduino IDE.

```
...
WiFi connected
IP address: 
192.168.1.13
Ready
IPess: 192.168.1.13
Attempting MQTT connection...connected
Setting LEDs:
r: 0, g: 0, b: 0
{"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"humidity":"0.00","motion":"","ldr":"0","temperature":"0.00","heatIndex":"nan"}
{"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"humidity":"0.00","motion":"standby","ldr":"0","temperature":"0.00","heatIndex":"nan"}
{"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"humidity":"0.00","motion":"standby","ldr":"60","temperature":"0.00","heatIndex":"nan"}
{"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"humidity":"0.00","motion":"standby","ldr":"60","temperature":"21.60","heatIndex":"nan"}
{"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"humidity":"67.70","motion":"standby","ldr":"60","temperature":"21.60","heatIndex":"16.64"}
```

ravensteijn, 2017-10-27:
Never mind. I now use this source.
https://github.com/ShawnCorey/ESP-MQTT-Multisensor-Authdiscovery

closed 2017-10-27 by ravensteijn

redparchel, 2017-11-28:
just wanted to follow up on this in case anyone else has this issue. as I had a similar issue and found my issue to be that I hadn't modified the PubSubClient.h from the PubSubClient library 
modifying the MQTT_MAX_PACKET_SIZE to be '512' as mentioned in Ben's video: [https://youtu.be/jpjfVc-9IrQ?t=10m42s](https://youtu.be/jpjfVc-9IrQ?t=10m42s)

from:
`#define MQTT_MAX_PACKET_SIZE 128`
to:
`#define MQTT_MAX_PACKET_SIZE 512`