bruhautomation/ESP-MQTT-JSON-Multisensor #48: LED flickering when motion is detected

markusressel, 2017-11-30:
Hi there,

great piece of code already. I have a small issue that I would like to fix (if possible).
First a little bit about my setup:

I use a single NodeMCU with all of the sensors and an RGB LED attached.
I use a dedicated MQTT server that is connected with the NodeMCU and Home Assistant.
In Home Assistant I have an automation set up that turns on the LED of the NodeMCU using the MQTT method and another automation that turns it off when there is no motion anymore.

At first I had an issue with "flickering states" in Home Assistant that would constantly trigger the "on" and "off" automation even when there was movement at all times.

[image]

I was able to fix this by introducing a small timeout before setting the ```pirValue``` to "standby" or ```pirState = 2``` like this:

```
int pirValue;
int pirStatus;
int lastTimeMotionDetected = 0;
String motionStatus = "standby";

[...]

pirValue = digitalRead(PIRPIN); //read state of the

if (pirValue == HIGH) {
  lastTimeMotionDetected = millis();
  
  if (pirStatus != 2) {
    motionStatus = "motion_detected";
    sendState();
    pirStatus = 2;
  }
} 
// add a timeout of 5 second before switching to standby to prevent "state flickering" in home assistant when motion is detected
else if (pirValue == LOW && pirStatus != 1 && (millis() > (lastTimeMotionDetected + 5000))) {
  motionStatus = "standby";
  sendState();
  pirStatus = 1;
}
```

**However** even though the LED is turned on at all times and Home Assistant doesn't turn it on and off constantly the LED itself flickers in a seemingly random interval of 1-2 seconds while movement is detected. At first I thought this was due to the fact that Home Assistant was switching states but now that Home Assistant doesn't do this anymore I'm kinda out of ideas.

Does someone know why this happens and how to fix this?

Home Assistant is not turning it on and off:
[image]
(Yes I was actually standing there waving my arms for 2 minutes... xD)

rdbahm, 2017-12-08:
Does the issue still occur if you set it to 100% white? I'm wondering if the AnalogWrite is being interrupted by the main loop, which would cause visible flickering. If it's set to 100% white, it shouldn't flicker if this is the case.

I also assume you're not using the "transition" feature to fade between "off" and "on" states?

markusressel, 2017-12-08:
I'm not sure if I understand what you mean by that.
My Home Assistant automation looks like this (there is no color specified):
```
- alias: 'NodeMCU1: Motion detected'
  hide_entity: True
  trigger:
    - platform: state
      entity_id: binary_sensor.nodemcu_1_pir
      from: 'off'
      to: 'on'
  action:
    - service: homeassistant.turn_on
      entity_id: light.nodemcu_1_led
      
- alias: 'NodeMCU1: No motion anymore'
  hide_entity: True
  trigger:
    - platform: state
      entity_id: binary_sensor.nodemcu_1_pir
      from: 'on'
      to: 'off'
  action:
    - service: homeassistant.turn_off
      entity_id: light.nodemcu_1_led
```

This is the definition of the LED in Home Assistant:
```
# Multisensor 1
- platform: mqtt_json
  name: "NodeMCU_1 LED"
  state_topic: "/home/multisensor/NodeMCU_1"
  command_topic: "/home/multisensor/NodeMCU_1/set"
  brightness: true
  flash: true
  rgb: true
  optimistic: false
  qos: 0
```

Should I turn off the "flash" option to prevent this? Or what exactly is this option used for?

I don't use the "transition" feature currently. Maybe I should check this out and see if it helps.

rdbahm, 2017-12-08:
Try just disabling the "flash" option - what "flash" does is temporarily
override the current light color with the requested light color - would
almost definitely cause flickering.

I asked about "transition" because the codepath for handling transitions is
totally separate from normal processing and might be somewhat more
vulnerable to problems like this.

markusressel, 2017-12-12:
Thanks for your help.

I turned off the "flash" option now. I'm not quite sure though if this really was the issue as I couldn't reproduce the flickering in the last days (hence the delay in response). I will close this issue though as it seems to be fixed.

closed 2017-12-12 by markusressel