bruhautomation/ESP-MQTT-JSON-Multisensor #51: Can't connect to MQTT Server?

TerryLansdown, 2018-03-30:
Hi, thanks for your efforts on the sensor, very cool.

I get all the way through to flashing the board and looking at the serial port. It connects to my WIFI, but then won't connect to my MQTT server. I think the error is a 'can't establish a TCP connection'. I can ping the nodeMCU and it's live on my network and other devices are interacting with the Mosquitto server normally. 

Any ideas? 

I'm not using any internal security on the MQTT server, and it's set to allow anonymous connections.

The serial port reports:
```
calibrating sensor Starting Node named sensornode01

Connecting to ********
.......................................................................................................
WiFi connected
IP address: 
192.168.0.106
Ready
IPess: 192.168.0.106
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
Attempting MQTT connection...failed, rc=-2 try again in 5 seconds
```

TerryLansdown, 2018-03-30:
Fixed it, I'd dumbly put 'http://' before my MQTT Server's IP.

Gah!

closed 2018-03-30 by TerryLansdown

MischaI, 2018-09-21:
http:// doesnt help. I have steel "Attempting MQTT connection...failed, rc=-2 try again in 5 seconds"