bruhautomation/ESP-MQTT-JSON-Multisensor #52: wdt timeout

jhavens12, 2018-04-05:
Is anyone else seeing the WDT timeout issues? For me, the sensor will work for a few days without issue and then it will drop offline. If you look at the console, it will show "wdt timeout ()" and freeze up until its physically unplugged and replugged in. Getting this with a few  ESP8266 boards. Anyone have any ideas?

mattlward, 2018-04-06:
I do not.. I have 3 sensing Wemos D1 mini's running espeasy and passing 2 values back to mqtt on a very regular basis.  A couple of them also have switches and they always respond.  I am running the Mosquito broker on Hassio.

mr-sneezy, 2018-04-26:
Do you have to have console running before the WDT, or is it readable after ?
Might check mine for this...

jhavens12, 2018-05-24:
Yes I've seen the WDT timeout when connected to the console, there is nothing after the wdt timeout line, just blank until the board is unplugged/replugged

SealedJoy, 2018-06-21:
Same problem here with wemos d1 mini