bruhautomation/ESP-MQTT-JSON-Multisensor #53: Add function declarations

wayne47git, 2018-04-17:
I'm not real familiar with **this** development environment but I've been hacking C code for over 40 years and doing embedded systems for 35+. 

I've been using Platformio to program my Sonoff devices so used that to build the Multisensor code as well. I needed to declare all the functions to make it compile (and, really, it's just a "good idea" to do that anyway). Suggest adding them:

```void setup_wifi();
void callback(char*, byte*, unsigned int);
void reconnect();
bool processJson(char* message);
void setColor(int inR, int inG, int inB);
int calculateVal(int step, int val, int i);
int calculateStep(int prevValue, int endValue);
void loop();
float calculateHeatIndex(float humidity, float temp);
void software_Reset() ;```

mr-sneezy, 2018-04-26:
I've been told that too by better coders than me.
Is that going to help code stability or just speed up compilation ?

consolacion, 2018-07-14:
it doesn't do anything for code stability.
It was necessary til I think IDE 1.6.7 if the functions were in a separate file, but no longer necessary now