bruhautomation/ESP-MQTT-JSON-Multisensor #55: Soil moisture sensor

manojjenago, 2018-04-20:
A request can you please add to read the data from a soil Moisture Water Sensor (https://www.amazon.com/gp/product/B01FDGUHBM/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1) Planning to add for automate the watering plants based on moisture..

MJ

manojjenago, 2018-04-26:
Please let me know , if we can add the soil moisture sensor to it??

mattlward, 2018-04-30:
I have these working on the digial I/O.  So a 1 or 0 depending upon water presence.  I have not tried them in an analog fashion yet.  But, I am not looking at soil moisture, but I am looking at basement moisture for a failed sump pump.

consolacion, 2018-07-14:
Ideally you would connect it to A0, but that's taken. The module you linked to has a digital output, but that only gives you a Boolean level indication. If that suffices, you can connect it to a free pin, otherwise you may need to ditch the LDR.
If you really need both, get a cheap PCF8591 ADC->I2C board