bruhautomation/ESP-MQTT-JSON-Multisensor #58: Constant ESP reboots

boardyuk, 2018-06-12:
Hi,

I've just setup 1 of these nodes to test as I want to have motion detection for some of the rooms in my house to control the lights with Home Assistant, however I'm experiencing a constant reboot of the ESP which is a little strange. I've got the ESP connected to my PC via USB to watch the serial monitor for the time being to ensure all messages are getting to MQTT.

Every few minutes (no specific timeout) the ESP is rebooting itself. I've got all the sensors connected except the RGB LED as I'm still waiting for those to arrive and I'm also undecided if I want the LED on all the nodes. The error I'm getting is -

`rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld`

I've had a look and it seems to be related to a reset pin, however I can't really see what's going on. Can anyone help me please?

Thanks!

mr-sneezy, 2018-06-12:
I had similar bugs. 
One thing that helped I think was to follow this information below in the link and add the pin assignments that were missed in the original code. My RGB led now behaves as expected and I seem to have got rid of most if not all random resets at the same time...
https://github.com/bruhautomation/ESP-MQTT-JSON-Multisensor/issues/38
Also there is a branch that has the 'random -ve sensor value data after reset' fixed, but I could not find it right now...

simonfell, 2018-06-25:
FWIW i have now found that the 2 sensors that i have built and configured seem to interfere with each other causing them to constantly reboot.  If i disconnect 1 the other is great.  I am guessing there is a simple solution, but i am struggling to see it at the moment.  
Any ideas???

Si

consolacion, 2018-07-14:
When you built the 2nd one, did you give it its own mqtt client name?

boardyuk, 2018-08-09:
I haven't yet found a solution to this. I've actually ripped out the RGB LED code from the sketch as I have no need for the LED and it still reboots.

I'm lost as to what is causing this really.

danielgonzacas, 2018-12-26:
you can try upload the sketch selecting Arduino/tools/erase flash/ sketch + wifi settings, maybe it can help you.