bruhautomation/ESP-MQTT-JSON-Multisensor #63: Error parsing value: 'dict object' has no attribute 'temperature'

mr-sneezy, 2018-08-10:
Hey guys, 
Since reflashing the multisensors with DanielXYZ2000 code revision I've not had any more problems with rubbish 0 values in the HomeAssistant sensor charts. However I do still get these HA internal log errors each time a multisensor either restarts itself, or maybe after a long period with little sensor change (sometimes not for days, then sometimes twice a day, it's very random).
While the 'Error parsing' messages are not stopping anything working, they do distract me from see more crucial info in the logs. 
I'd like to eliminate them. 

Any of you know why HA still sees a problem with the DHT22 part of the MQTT  ?

This is what I get in the logs in a block randomly, alway for the DHT22 related values...
`018-08-10 17:18:17 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'humidity' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"standby","ldr":"0"}, template: {{ value_json.humidity | round(1) }})
2018-08-10 17:18:17 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'heatIndex' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"standby","ldr":"0"}, template: {{ value_json.heatIndex | round(1) }})
2018-08-10 17:18:17 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'temperature' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"standby","ldr":"0"}, template: {{ value_json.temperature | round(1) }})
2018-08-10 17:18:17 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'humidity' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"standby","ldr":"0"}, template: {{ value_json.humidity | round(1) }})
2018-08-10 17:18:17 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'heatIndex' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"standby","ldr":"0"}, template: {{ value_json.heatIndex | round(1) }})
2018-08-10 17:18:17 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'temperature' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"standby","ldr":"0"}, template: {{ value_json.temperature | round(1) }})
2018-08-10 17:18:17 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'humidity' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"standby","ldr":"0"}, template: {{ value_json.humidity | round(1) }})
2018-08-10 17:18:17 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'heatIndex' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"standby","ldr":"0"}, template: {{ value_json.heatIndex | round(1) }})
2018-08-10 17:18:17 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'temperature' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"standby","ldr":"0"}, template: {{ value_json.temperature | round(1) }})
2018-08-10 17:22:40 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'humidity' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"standby","ldr":"0"}, template: {{ value_json.humidity | round(1) }})
2018-08-10 17:22:40 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'heatIndex' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"standby","ldr":"0"}, template: {{ value_json.heatIndex | round(1) }})
2018-08-10 17:22:40 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'temperature' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"standby","ldr":"0"}, template: {{ value_json.temperature | round(1) }})
2018-08-10 17:22:40 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'humidity' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"motion detected","ldr":"0"}, template: {{ value_json.humidity | round(1) }})
2018-08-10 17:22:40 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'heatIndex' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"motion detected","ldr":"0"}, template: {{ value_json.heatIndex | round(1) }})
2018-08-10 17:22:40 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'temperature' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"motion detected","ldr":"0"}, template: {{ value_json.temperature | round(1) }})
2018-08-10 17:22:40 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'humidity' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"motion detected","ldr":"0"}, template: {{ value_json.humidity | round(1) }})
2018-08-10 17:22:40 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'heatIndex' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"motion detected","ldr":"0"}, template: {{ value_json.heatIndex | round(1) }})
2018-08-10 17:22:40 ERROR (MainThread) [homeassistant.helpers.template] Error parsing value: 'dict object' has no attribute 'temperature' (value: {"state":"OFF","color":{"r":255,"g":255,"b":255},"brightness":255,"motion":"motion detected","ldr":"0"}, template: {{ value_json.temperature | round(1) }})`

mkomarac, 2018-10-19:
I have the same issue. have you manage to solve it?

mr-sneezy, 2018-10-20:
Nope. I have  V79.3 of HA and still get these errors. I think every time the SensorNode resets HA throws this list, and the SensorNode resets itself a couple of times a day on average.

tubalainen, 2018-11-18:
Yes, Ive got the same issue here as well.