bruhautomation/ESP-MQTT-JSON-Multisensor #67: Not declared buffer size while compiling

mcsabi, 2019-01-01:
Any thoughts would be appreciated why I am receiving this error when compiling the code in Arduino:

Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (3M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

C:\Users\xxxx\Documents\Arduino\Multisensor_Bruh\sketch_jan01a\sketch_jan01a.ino: In function 'bool processJson(char*)':

sketch_jan01a:260: error: 'StaticJsonBuffer' was not declared in this scope

   StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;

   ^

sketch_jan01a:260: error: 'jsonBuffer' was not declared in this scope

   StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;

                                 ^

sketch_jan01a:264: error: 'ArduinoJson::JsonObject' has no member named 'success'

   if (!root.success()) {

             ^

C:\Users\xxxx\Documents\Arduino\Multisensor_Bruh\sketch_jan01a\sketch_jan01a.ino: In function 'void sendState()':

sketch_jan01a:335: error: 'StaticJsonBuffer' was not declared in this scope

   StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;

   ^

sketch_jan01a:335: error: 'jsonBuffer' was not declared in this scope

   StaticJsonBuffer<BUFFER_SIZE> jsonBuffer;

                                 ^

sketch_jan01a:340: error: invalid initialization of non-const reference of type 'ArduinoJson::JsonObject& {aka ArduinoJson670_0_0::ObjectRef&}' from an rvalue of type 'ArduinoJson670_0_0::ObjectRef'

   JsonObject& color = root.createNestedObject("color");

                                                      ^

sketch_jan01a:354: error: 'ArduinoJson::JsonObject' has no member named 'measureLength'

   char buffer[root.measureLength() + 1];

                    ^

sketch_jan01a:355: error: 'ArduinoJson::JsonObject' has no member named 'printTo'

   root.printTo(buffer, sizeof(buffer));

        ^

sketch_jan01a:355: error: 'buffer' was not declared in this scope

   root.printTo(buffer, sizeof(buffer));

                ^

exit status 1
'StaticJsonBuffer' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

per1234, 2019-01-02:
This error means you're using a version of ArduinoJson library that is not compatible with this code. The ArduinoJson library 6.x.x versions have breaking changes that make them not not work with code written for the ArduinoJson 5.x.x versions.

The recommended solution is to roll back to the newest 5.x.x release of ArduinoJson:
1. Sketch > Include Library > Manage Libraries...
1. Wait for the download to finish.
1. In the "Filter your search..." box, type "arduinojson".
1. In the search results, click on "ArduinoJson by Benoit Blanchon"
1. From the dropdown version menu, select "5.13.4".
1. Click "Install".
1. Wait for the installation to finish.
1. Click "Close'. The code should now compile.

The alternative solution is to update the code to work with the 6.x.x versions of ArduinoJson. You will find a guide to this here:
https://arduinojson.org/v6/doc/upgrade/

Reference:
https://arduinojson.org/v5/faq/error-jsonbuffer-was-not-declared-in-this-scope/

mcsabi, 2019-01-02:
Thanks a lot works like a charm with 5.13.4