bruhautomation/ESP-MQTT-JSON-Multisensor #72: Temp reading at least 15 degrees low.

ebuford, 2019-07-31:
I have built the Multisensor with a DHT22 and NodeMCU along with the TEMT6000 & AM312 PIR.
I'm using the same Pinouts as the Multisensor Video shows. 
I've flashed this a couple of times using the exact code straight out of this site and Each time it comes up reading at least 15 degrees F low. Also the humidity is low as well. 
This lead me to believe that perhaps the DHT22 was bad. So I set out to test it. 
I then set up the DHT22 and flashed it with this code https://simple-circuit.com/esp-12e-nodemcu-ssd1306-dht22-am2302/ 
And the temp and humidity was correct. 
So I have returned to the original code,  trying to figure out what might be causing this to run so low. 
I've read thorough the issues but it seems like all the issues with temp have been caused by heat build up in a case - so far these have not been in a case and are just laying out on my bench. 
The only lines I have changed in the code are SSID & Password along with the MQTT info.
Any help would be appreciated.

RedDevilAustria, 2019-08-30:
Hi I have actually the same issue ... Does anyone have an idea?

ebuford, 2019-08-30:
@RedDevilAustria I researched this a lot and didn't get an answer. 
I've played with the DHT22 a fair amount have deployed several around my place; but I just couldn't get them to work the Multisensor from BRUH Automation. 
Recently Digiblur did a great video on building one very similar to the BRUH sensor so I made one without the LED (which I'm not using anyway) and it works great.
Here's the link to that one. 
https://www.youtube.com/watch?v=9Yu57vjz7AY
Hopefully you're question will get more answers than mine did.

RedDevilAustria, 2019-08-30:
@ebuford : Thank you very much for the video - I'll give it a try - maybe the sensor works at some point - never give up coding ! :) Should I find something I'll share it

RedDevilAustria, 2019-09-01:
@ebuford: got it working now - inserted in the setup function after serial begin
-->dht.begin();

and in the loop function before reading the new temperature
--> delay(3500);
    float newTempValue = dht.readTemperature(IsFahrenheit);
    float newHumValue = dht.readHumidity();

this works fine for me now and reads super accurate. Hope this one works also for you!

ebuford, 2019-09-06:
@RedDevilAustria I have updated my code and will give it a try this weekend. 
Thanks for sharing!